ziirish/OctoPrint-TuyaSmartplug #2: Feature request - Add USB power toggle support

GhostlyCrowd, 2018-11-21:
My Smart switch works great with your plugin, however i cannot control the usb ports on my switch can you add support for "switch_usb1" and "switch_usb2"  I can test this for you if you like.

ziirish, 2018-11-21:
Hello,

Sure, I just need the full status of your plug. You can find it in the `.octoprint/logs/plugin_tuyasmartplug_debug.log` file.

Something like that:

```
[2018-11-21 20:36:09,220] DEBUG: Status: {u'devId': u'04110747600194XXXXXX', u'dps': {u'1': True, u'11': 0}}
```

ziirish, 2018-11-21:
Ok perfect, thanks.

Actually this has already been implemented in #1.
I haven't release a new version yet though since it needs some polish, but if you know what you are doing, then you can download the master archive and install the plugin through it.

GhostlyCrowd, 2018-11-21:
I'll give it a shot after I finish this print, thanks!

GhostlyCrowd, 2018-11-21:
alright @ziirish  i installed hte master. I still can only toggle the ""switch_1" and ""switch_2" but not "switch_usb1" and "switch_usb2" 

I'm getting these switch names from the same xml dump that i got my keys from.

ziirish, 2018-11-21:
To give you some more context/detail, there are two main changes introduced by this request:

1. You must provide a label for each of your switch. This label must be unique across all your plugs/switches.
2. You can now choose the switch slot a given label will act on.

3. The `@TUYAON` and `M80` gcodes can take the label as parameter. If a IP is given, then the first matching entry will be returned

ziirish, 2018-11-21:
ok, so now you have a new field in the setting panel called "Plug Slot". Regarding the log you provided me, you can set slot 1 to 4.

Just keep in mind you have to provide a unique label for each entry.

GhostlyCrowd, 2018-11-21:
Right now i can turn on both the 110V outlets seperatly. but the USB ports i cannot. ive tried naming them switch 3 and 4 respectivly and they dont respond.  

I actually had the "plug slot" option from the plugin install through the plugin manager in octoprint, however I have uninstalled that one and installed the master zip. same symptoms though i cannot control the usb plugs only plus "1 and 2"

ziirish, 2018-11-21:
hm sorry, the slots are 7 and 8 regarding your logs.

GhostlyCrowd, 2018-11-21:
Well that makes sense I was just reading them as you replied. I assumed it was 1-4 

And it works! Thanks.

closed 2018-11-21 by GhostlyCrowd