ziirish/OctoPrint-TuyaSmartplug #3: Gcode Trigger behavior

therealpanse, 2019-02-12:
Hi, just a quick question about the usage of the Gcode triggers.

I know I have to use M80 [IP], but how would I do that, when the Printer is turned off? While it's not powered on, the Octoprint Terminal is inactive and won't accept the M80 command at all.

I'm trying to remotely turn on the printer via Printoid or OctoRemote on my phone in case my PC isn't available. Is there some system command I could use?

ziirish, 2019-02-12:
Hi,

Well, yes, the GCode trigger requires your printer to be turned-on (or at least its controller board).
Depending on your printer having the USB cable plugged into your octoprint device is enought to only power the controller board, you'll need the main power supply only to heat-up your printer and control your steppers.

You could otherwise probably call the octoprint API to send the power-on signal to your plug through my plugin, unfortunately I don't know yet if this is doable and how that would be doable (This plugin is actually based on the work done by @jneilliii [here](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug))

I'll let you know when I have some time to check this (hopefully next week).

ziirish, 2019-07-18:
Hello,

Sorry it's been a loooooong time...
I'm not using this plugin anymore, but I did some research and I'm pretty sure you can craft a special request (curl) in order to switch on your smartplug and thus your printer.

You would run something like:

```
curl -H "Content-Type: application/json" -H "X-Api-Key: xxxxxxx" -X POST -d '{"command": "turnOn", "label": "TevoPlug"}' http://your.octo.print/api/plugin/tuyasmartplug
```

Where `TevoPlug` is your plug label as set in your preferences and `xxxxxxx` is your OctoPrint API Key.

ziirish, 2019-11-06:
I'm now closing this issue as it as been replied in my view.

closed 2019-11-06 by ziirish