ziirish/OctoPrint-TuyaSmartplug #4: Almost there -- can't trigger on/off when Octopi is on/offline

acraigl, 2019-07-24:
I was able to successfully configure my Tuya smart plug and can toggle it on/off from the Octoprint menu header. Where I'm getting stuck is controlling it through events. Like turning it off when the server is shut down, or when a print completes.

I was able to pass the command M81 10.0.0.xxx through the terminal and shut off the plug, so I know I'm close...

I've set the plugin to the following, and GCODE trigger is checked
`Run System Command Before Off: M81 10.0.0.xxx`

I also have the GCODE script "After print job completes" set to M81 10.0.0.xxx. I'm sure I'm missing something obvious #noob. Thanks!

ziirish, 2019-07-24:
I guess it depends on the kind of plug you are using (does it have several slots?) or if you setup several plugs.
Anyway, the GCode trigger should also work against labels as well.
So instead of using `M81 10.0.0.xxx` you can instead use `M81 myPlugLabel`

acraigl, 2019-07-24:
@ziirish thanks for the quick reply! It's a simple one-plug socket. 
https://www.amazon.com/Jinvoo-Wireless-Outlet-Schedule-Control/dp/B07121PYWQ

I'll try the label option and see if that gets me any further. Thanks!

ziirish, 2019-07-24:
You can also enable debug logging and check the logs that are located here: `~/.octoprint/logs/plugin_tuyasmartplug_debug.log` and see if you find anything useful.

acraigl, 2019-07-25:
I added in the printershutdown plugin to the mix and it successfully turned off when I completed a test print, so I'm happy with everything right now. Thanks for the help!

ziirish, 2019-07-26:
Oh yes, sorry. I thought you were already using it!

I'm closing this now if it's ok for you.

closed 2019-07-26 by ziirish

acraigl, 2019-08-06:
> Oh yes, sorry. I thought you were already using it!
> 
> I'm closing this now if it's ok for you.

@ziirish That's fine... thanks again. 
FWIW, I just updated the firmware on my Ender 3 -- moved from factory 1.1.6  to the TH3D Unified Firmware Package (based on 1.1.9) -- and all the other things just started working. For example, when I power on the Pi manually or issue a server shutdown command from the web console, the Tuya plug responds by turning on and turning off respectively. 

Might be worth noting that some of the behavior is highly dependent on the firmware that the printer is running.