ziirish/OctoPrint-TuyaSmartplug #5: No longer working?

jfrancais, 2019-09-04:
Recently this plugin stopped working for me.  I'm getting errors in octoprint when I attempt to cycle the power.  I confirmed my Device ID and local key are correct and I can power cycle the plug externally with the app so I know communication is working and the ip address is correct.

Is anyone else having this issue?

ziirish, 2019-09-05:
Could you please enable debug logging and send me back what's happening in the logs?

There is a `enable debug logging` checkbox in the settings panel. Then the logs are located here: `~/.octoprint/logs/plugin_tuyasmartplug_debug.log`.

Thanks

jfrancais, 2019-09-05:
Jared Francais | tel 306.975.3259
Programmer Analyst
City of Saskatoon | 222 3rd Avenue North | Saskatoon, SK  S7K 0J5
jared.francais@saskatoon.ca<mailto:jared.francais@saskatoon.ca>
www.saskatoon.ca<http://www.saskatoon.ca/>
Follow us on Twitter or Facebook.

If you receive this email in error, please do not review, distribute or copy the information.
Please contact the sender and delete the message and any attachments.

ziirish, 2019-09-05:
I'm not using those plugs anymore so I won't be able to try this at the moment. If others are experiencing the same issue, please let us know.

Now, the plugin seem to still be able to request the plug status.

What I know, is those plugs only support ONE client at the time to interact with them.
Could you make sure your phone/tablet/whatever has the app installed are turned off (just switch the wifi off) and then try again?

I implemented a retry mechanism but the error returned seem different than what it used to be (maybe there was a firmware update or something).
If the above test fixes your issue, then I'll implement the retry mechanism no matter what error is returned.

jfrancais, 2019-09-06:
I tried again, no other clients connected to the plugin (the client was only used for initial setup and then to troubleshoot after failures started)

ziirish, 2019-09-06:
Thanks for the heads up.
I'll try to give it a look when I have a chance.

stickyblue, 2019-09-10:
I've exactly the same issue. He's not the only one. But you seems to be on it, let me know if you have any news :p

tuxfamily, 2019-09-19:
Hello. After digging hours, for me the issue was the outdated version of "pytuya" installed AND the protocol version used. So after manually updating pytuya with the github version and forced the use of protocol version 3.3, everything is back to normal ... Hope it helps.

jfrancais, 2019-09-19:
can you provide instructions tuxfamily?

ziirish, 2019-09-19:
@tuxfamily Thanks for digging into this and thanks for sharing your findings!
@jfrancais I will plan a plugin update to embed the latest pytuya version. There are no up-to-date packages of pytuya we can rely on so I will have to *statically* integrate it with the plugin somehow.

Sorry for the delay, but be assured I keep this in my todo.

ziirish, 2019-09-20:
So, I just gave a try at the current plugin and it works fine.

@tuxfamily could you elaborate a bit more on what you did exactly?
How do you force protocol 3.3?

Is this what you did?

```python
d = pytuya.OutletDevice('...')
d.version = 3.3
```

Is this the only change required?

tuxfamily, 2019-09-24:
Sorry for the late reply.

First of all, a quick precision, I'm using OctoPi, so paths may differ from other distribs.

My "fix" is not really a solution as it is really dirty.

Firstly, I replaced `__init__.py` in `~/oprint/lib/python2.7/site-packages/pytuya` with the latest version found here [clach04/python-tuya](https://github.com/clach04/python-tuya/blob/master/pytuya/__init__.py)

And next, I edited this file and set the default protocol to `3.3` instead `3.1` (line 160)
```diff
160c160
<         self.version = 3.1
---
>         self.version = 3.3
```

And that's all.
Not clean at all, but it works (until next updates I guess) :)

PS: another precision; the target device is a brand new socket bought few days ago in a local store named "Action" which sale products branded "LSC Smart Connect" and I did not upgraded it with the Tuya app, so it is shipped with this firmware / new protocol by default.

wieltar, 2019-10-15:
> 
> 
> Sorry for the late reply.
> 
> First of all, a quick precision, I'm using OctoPi, so paths and versions may differ from other distribs.
> 
> My "fix" is not really a solution as it is really dirty but it consists to manually update python-tuya to the latest version and to force the usage of the protocol version 3.3.
> 
> So, firstly, I replaced `__init__.py` in `~/oprint/lib/python2.7/site-packages/pytuya` with the latest version found here [clach04/python-tuya](https://github.com/clach04/python-tuya/blob/master/pytuya/__init__.py)
> 
> And next, I edited this file and set the default protocol to `3.3` instead `3.1` (line 160)
> 
> ```diff
> 160c160
> <         self.version = 3.1
> ---
> >         self.version = 3.3
> ```
> 
> And that's all.
> Not clean at all, but it works (until next updates I guess) :)
> 
> Of course, an option to choose the protocol version to use at device setup would be better ... but for that to works, python-tuya must be to the latest version :/
> 
> PS: another precision; the target device is a brand new socket bought few days ago in a local store named "Action" which sale products branded "LSC Smart Connect" and I did not upgraded it with the Tuya app, so it is shipped with this firmware / new protocol by default.

What file do you mean by "this file"? I am trying to use my woox plug (installed via the Tuya app) but i too discovered this plugin doesn't work

tuxfamily, 2019-10-17:
@wieltar sorry if I was not clear
By `this file` I mean `~/oprint/lib/python2.7/site-packages/pytuya/__init__.py` but not the original one, the latest version that I took [here](https://github.com/clach04/python-tuya/blob/master/pytuya/__init__.py) 

To summarize, the problems are that:
- the version of python-tuya shipped with octopi (for me) is outdated, so it does not have the changes for the protocol version 3.3
- the plugin does not offer the possibility to choose the protocol version to use 
- AND, the default version used in python-tuya is the 3.1, even with the latest version

SO, I manually "update" python-tuya, and I edit it to force the use of protocol version 3.3.

closed 2019-10-21 by ziirish

ziirish, 2019-10-21:
Hello,

I have just released a new version with updated pytuya version and support for v3.3 protocol.
You'll then have a new setting option to choose whether you want to use this 3.3 protocol or not.

Please note that the plugin settings **WILL** be reset during the update process as the settings structure change.

commit 1f06290d (2019-10-21): support newer 3.3 protocol (fix #5)