ziirish/OctoPrint-TuyaSmartplug #6: Can't install the plugin

Aure42, 2019-11-05:
Hi when i try to install the plugin i have this : 

```
    ERROR: Command errored out with exit status 1:
     command: /home/pi/oprint/bin/python2 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-cD0Kvk/pytuya/setup.py'"'"'; __file__='"'"'/tmp/pip-install-cD0Kvk/pytuya/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-install-cD0Kvk/pytuya/pip-egg-info
         cwd: /tmp/pip-install-cD0Kvk/pytuya/
    Complete output (7 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-cD0Kvk/pytuya/setup.py", line 9, in <module>
        import pytuya
      File "pytuya/__init__.py", line 28, in <module>
        import pyaes  # https://github.com/ricmoo/pyaes
    ImportError: No module named pyaes
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

ziirish, 2019-11-05:
Thanks, I'll have a look when I have a chance

ziirish, 2019-11-06:
I have just published a new version that should address your issue.

Aure42, 2019-11-06:
I'm still getting this : 
ERROR: Command errored out with exit status 1:
command: /home/pi/oprint/bin/python2 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-eSJF4_/pytuya/setup.py'"'"'; __file__='"'"'/tmp/pip-install-eSJF4_/pytuya/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-install-eSJF4_/pytuya/pip-egg-info
cwd: /tmp/pip-install-eSJF4_/pytuya/
Complete output (7 lines):
Traceback (most recent call last):
File "<string>", line 1, in <module>
File "/tmp/pip-install-eSJF4_/pytuya/setup.py", line 9, in <module>
import pytuya
File "pytuya/__init__.py", line 28, in <module>
import pyaes  # https://github.com/ricmoo/pyaes
ImportError: No module named pyaes
----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

Octoprint fresh install and update upgrade done

ziirish, 2019-11-06:
Well, this seem like a packaging issue with one of the plugin requirements.

Right now I can't find a way to fix it within the plugin.

What you can do as a workaround though is this:

1. ssh into your octopi: `ssh pi@your.octopi.ip`
2. then run the following: `source ~/oprint/bin/activate; pip install pyaes==1.6.1`

Finally try to install the plugin again through the plugin manager.

In the meantime I'm re-opening the issue in case I can find something better

Aure42, 2019-11-06:
Thanks you now it's ok. Good job, thanks for fast reply

rchauderlot, 2019-11-13:
I had same error. I managed to install it by executing the commented command:
source ~/oprint/bin/activate; pip install pyaes==1.6.1
But without executing that pior to the installation the plugin does not install.

ziirish, 2020-01-03:
Hello,

Just a quick note to let you know that I haven't figured a clean way to fix this yet.
Still thinking...

eightheads, 2020-01-04:
> Hello,
> 
> Just a quick note to let you know that I haven't figured a clean way to fix this yet.
> Still thinking...

Thanks for the hard work @ziirish !

ziirish, 2020-01-05:
I have proposed a fix upstream since I believe this has to be fixed there.

ziirish, 2020-01-18:
For the record, my proposal to fix `python-tuya` has been accepted and merged, so hopefully the issue will soon be resolved.

Deses, 2020-02-13:
Just so you know, I still had to run `source ~/oprint/bin/activate; pip install pyaes==1.6.1` to install your plugin.

You command worked great, though!

ziirish, 2020-02-14:
Yes, although the patch has been merged upstream, we still need `python-tuya` to release a new version that embed the change.

That's why I'm keeping this issue open until it is definitely fixed.

ILIDAN-SG, 2020-02-21:
> I had same error. I managed to install it by executing the commented command:
> source ~/oprint/bin/activate; pip install pyaes==1.6.1
> But without executing that pior to the installation the plugin does not install.

That works for me, thank you @ziirish. If octoprint is installed in raspbian, the code must be:
`source /home/pi/OctoPrint/venv/bin/activate; pip install pyaes==1.6.1`

Rebel154, 2020-02-27:
I've just tried to install the plugin on my Raspberry Pi3B+ and keep getting the same error as in the opening post. I tried running the command as listed by ILIDAN-SG above but it made no difference.

closed 2020-05-16 by ziirish

jrowen, 2020-05-18:
I'm still seeing this error when installing today
```
Installing plugin "OctoPrint-TuyaSmartplug" from https://github.com/ziirish/OctoPrint-TuyaSmartplug/archive/master.zip...
/home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install https://github.com/ziirish/OctoPrint-TuyaSmartplug/archive/master.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/ziirish/OctoPrint-TuyaSmartplug/archive/master.zip
Downloading https://github.com/ziirish/OctoPrint-TuyaSmartplug/archive/master.zip
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-TuyaSmartplug==0.2.3) (1.4.0)
Collecting pyaes==1.6.1
Downloading https://files.pythonhosted.org/packages/44/66/2c17bae31c906613795711fc78045c285048168919ace2220daa372c7d72/pyaes-1.6.1.tar.gz
Collecting pytuya==7.0.6
Downloading https://files.pythonhosted.org/packages/35/82/a10d977c4a0879c37275e2793d983a8b7ac0f6738e937cdc159d4f3a15c1/pytuya-7.0.6.zip
ERROR: Command errored out with exit status 1:
command: /home/pi/oprint/bin/python2 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-5V9t9_/pytuya/setup.py'"'"'; __file__='"'"'/tmp/pip-install-5V9t9_/pytuya/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-install-5V9t9_/pytuya/pip-egg-info
cwd: /tmp/pip-install-5V9t9_/pytuya/
Complete output (7 lines):
Traceback (most recent call last):
File "<string>", line 1, in <module>
File "/tmp/pip-install-5V9t9_/pytuya/setup.py", line 9, in <module>
from pytuya.const import __author__, __version__
File "pytuya/__init__.py", line 28, in <module>
import pyaes  # https://github.com/ricmoo/pyaes
ImportError: No module named pyaes
----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output
```

commit 6045339b (2019-11-06): fix: add missing requirement (fix #6)
commit 4c21879d (2020-05-16): fix: a new version of pytuya has been released and should fix #6 and #11