ziirish/OctoPrint-TuyaSmartplug #7: Some icons won't work in the Icon Class field

sonnaps, 2019-12-18:
I'm trying to get a fan icon to show.  I tried using the input of "icon-fan", just "fan", etc. but it keeps coming up blank.  Some of those icons from that fontawesome.com site work and some do not.  I ended up using "icon-asterisk" and that kinda looks like a fan, lol.   So far, "icon-bolt", "icon-asterisk", and "icon-lightbulb" all seem to work.  Is there something that I am missing in order to get a fan icon to show up?  I tried a bunch of them from that site, and it's pretty much hit or miss as far as which ones will work.

ziirish, 2020-01-03:
Hello,

I'll have to check further, but I think octoprint ships an old version of fontawesome.
Can you check [this icon list](https://fontawesome.com/v4.7.0/icons/) instead?

sonnaps, 2020-09-26:
You are correct, thanks so much.  Sorry I never closed this out.

closed 2020-09-26 by sonnaps