ziirish/OctoPrint-TuyaSmartplug #8: Local Key? What is that and where do I get it?

Deses, 2020-02-13:
I have searched high and low in the SmartLife app for a Local key but can't find one!

Is it the OctoPrint API key I can get from "Access control"?

Deses, 2020-02-13:
OK, I failed to see this on the Plugin page.

Deses, 2020-02-13:
Sorry 'bout that!

Deses, 2020-02-14:
I've been trying to get the Local Key for the past 3 hours but nothing works:

- The apps have been updated and they work different now, and the old versions no longer work.
- Capturing the packages with Packet Capture breaks the app and it only throws 50509 errors
- The files under `/data/data/com.tuya.smartlife/shared_prefs/` none seems to have the Key, and `preferences_global_key<someid>.xml` is an empty `map` tag.

I'm at my wits end... how am I supposed to get the LocalKey in 2020?

rathus, 2020-02-14:
Spent the day also struggling to set this up, I've got a step further then you, so let's see if you can take it one step further then me now.

The Packet Capture app can get the LocalKey from "Jinvoo Smart" version 1.03 from [APKpure](https://apkpure.com/jinvoo-smart/com.xenon.jinvoo).  As you found out, the other apps suggested don't even have versions old enough available, or if you can find the old versions, they can't connect, or they have that 50509 error.

Now I have the keys, but the plugin getting connection refused when trying to send the command.

Deses, 2020-02-14:
In my android 10 device (Oneplus 7 Pro) the version 1.0.3 of Jinvoo Smart is crashing as soon as I open it.

After a good night of sleep, I'm digging around more and [this helpful comment](https://github.com/codetheweb/tuyapi/issues/5#issuecomment-571237077) said that only 3.6.1 and earlier write the `preferences_global` xml correctly. Here's the APK: https://www.apkmirror.com/apk/tuya-inc/smart-life-smart-living/smart-life-smart-living-3-6-1-release/

I'm now trying to get the key and I'll report back in a minute. :)

rathus, 2020-02-14:
Oh, ignore my last comment, a search of an issue earlier yielded a tidbit from  the author.  One concurrent connection on the switches.  I force-closed the Jinvoo app, waited a minute, and now it works!  

Good luck with getting the LocalKey.   My phone is like 4-5 years old so maybe that's how i got away with the JinVoo 1.0.3 app.  Now 5AM is bedtime.

Deses, 2020-02-14:
I don't know why didn't I though about using my old phone running android 8.... damn! Maybe it works there!

ziirish, 2020-02-14:
Hello,

Yes sorry, I know those devices are a pain on that point (getting the 'Local Key').
I'm not using them anymore for that reason actually.

Last time I tried, I had to run tcpdump on my WiFi AP. I just gave it a try and I don't seem to be able to extract the key anymore from a fresh pcap.
I guess your best chance is to try an old app on an old device :/

Sorry for the inconvenience.

Deses, 2020-02-14:
> Oh, ignore my last comment, a search of an issue earlier yielded a tidbit from the author. One concurrent connection on the switches. I force-closed the Jinvoo app, waited a minute, and now it [...]

So what did you do? I haven the Jinvoo 1.0.3 app on my old phone and it works, and the Logcat shows stuff filtering by `D/Business ggg`... but I don't know what I'm looking for.

I'm pretty pissed off with the available [documentation](https://github.com/clach04/python-tuya/wiki) and how little specific it is. :(

rathus, 2020-02-14:
I didn't do any filtering, I just used the Packet Capture" app, set it to
monitor only the Jinvoo app, then ran Jinvoo and added my device.  The
catcher then had one larger capture 10-20kb or so, and in there i found a
JSON request with ~30 elements, right near the end was
"localKey":"xxxxxxxxxxxxxxxx" along with some other entries like my device
name, device ID, etc to confirm I had the right device.

Deses, 2020-02-14:
Oh my Goodness, finally! I got it with Packet Capture. Thank you, @rathus!!

VoarL, 2020-02-19:
Hey guys, I'm wondering how you got it working. I was able to get the device ID as well using an old android phone but I'm still getting a status unknown error.

Deses, 2020-02-19:
@VoarL The Device ID is something that any app will show in any app, that's not something hidden.
The LocalKey on the other hand... oh boy.

I managed to get it through an old Android 8.0 phone using Jinvoo 1.0.3 and Packet Capture to capture the data. Just go into Packet capture settings and install the certificate or you'll get error 50509 all the time.

Hope this helps!!!

VoarL, 2020-02-19:
@Deses So does the LocalKey ever change? I got it using Jinvoo 1.0.3 but I wasn't able to control the device using the app since it was asking me to update, if that matters. I'm sure I got the IP right since I assigned it myself, is there anything else I can check?

Deses, 2020-02-19:
LocalKey should not change, no.

I'm not really sure what you are trying to do, but you can uninstall the apps from the phone now and put the ID, LocalKey, IP, 3.3 protocol, etc to the OctoPrint plugin and you should be done with it.

If you want to still be able to control the plug through the app, I'd suggest uninstalling the old app and getting the latest SmartLife app and pair it just as you would normally do... you already got the Key after all! :)

VoarL, 2020-02-19:
Thanks but its still not working. I tried reinstalling this plugin and it still didn't work. Do you know what Plug Slot means? I did not see that in the documentation.

VoarL, 2020-02-19:
When I try changing that, I instantly get errors similar to when I enter the wrong IP address. Usually it takes a few seconds before the error pops up.

Deses, 2020-02-19:
The LocalKey you got was in an xml looking file that clearly said that the number was the LocalKey?
Maybe you don't have the key after all.

Also, have you checked the Version 3.3 protocol box?

[image]

VoarL, 2020-02-19:
I should add that the log file is filled with Connection refused and Network errors. Changing the Plug slot gave "something went wrong". I'm using Teckin plugs and I'm sure they're Tuya based.

VoarL, 2020-02-19:
Yeah, I got it using what you guys mentioned earlier with PacketCapture and it was like "localkey": ...

Deses, 2020-02-19:
That should be it, then! Double check the Device ID and use something like Angry IP Scanner to see if the plug is using the right IP. I had to wait until the DHCP lease expired to set the final IP.

VoarL, 2020-02-19:
I'm using the Device ID provided by the app.
Spent too much time on this, I guess this is a sign I should stop being lazy and use a relay.
If you have any ideas let me know. Thanks!

MrAwesome1996, 2020-02-23:
Funny, I switched to trying to use a smart plug because I tried using a relay _first_ and the startup current was enough that the contacts in the relay always welded closed (they were cheap relays though). 

This all was very helpful. I'd just like to add as an alternative for those viewing this in the future that I also found success using Smart Life 3.6.1 as @Deses mentioned. I used _BlueStacks_ (an android emulator, which I learned about [here](https://community.openhab.org/t/step-by-step-guide-for-adding-tuya-bulbs-wi-fi-smart-led-smart-life-app-to-oh2-using-tuya-mqtt-js-by-agentk/59371)) since I didn't want to root my phone. I set up my smart plug with the current Smart Life app on my phone, then installed Smart Life 3.6.1 on BlueStacks and logged in, viewed the plug's information and turned it on and off a couple times. Then I used a file explorer to go to `/data/data/com.tuya.smartlife/shared_prefs` and viewed the file `preferences_global_key<chars and nums>.xml` which, thankfully, featured the localKey and not just an empty `map` tag like the current version. I just did a search in the document for the word "key" to find it in the jumble of "&quot"s and letters and numbers in the file.

Deses, 2020-02-23:
Using an android emulator is a galaxy brain move that I'm sure no one else thought before! I'm definitely doing that next time so I don't have to mess with my old phone.

About the relays... That thing that happens to your relays is exactly why I didn't bother. Everyone is using that cheap Arduino relay rated for 10A and that's not nearly enough for a printer. I've read enough horror stories to realize that a relay is not a good idea.

Deses, 2020-02-25:
I Installed SmartLife app on my phone and paired the plug and it stopped working through the OctoPrint plugin... perhaps the LocalKey changed? 

Anyway, thanks to @MrAwesome1996 explanation of what he did, I'm back up and running again.

Just a heads up: Bluestacks doesn't have root enabled and you need it to access `/data/data/com.tuya.smartlife/shared_prefs`, so I had to use https://bstweaker.tk/ to root it.

Here's a good guide: https://sangams.com.np/how-to-root-bluestacks-latest-version/

capillonpub, 2020-07-13:
Hi,

I, ve got a TECKIN smart plug that is weel configured and working with smart life from tuya. I managed to get the IP, ID, and Key with an old phone and packet capture. But when i try with the tuya smartplug in Octopi, it doesn't work, saying 'Status unknown for 192.168.1.110. Double check IP Address\Hostname in tuyasmartplug Settings'.
I've checked that all Smartlife app are off on all other devices. I don't know what to do.
I've done all the process twice and retrieved the same key and id. the IP is manually assigned at router level and verified. I have uninstalled and installed again the plugin...
how to retrieve a log on octopi? 
How to check with another app the id and key?

Thanks for your help.
Philippe

ziirish, 2020-07-15:
Hello,

You could try checking the `Version 3.3` option.

capillonpub, 2020-07-15:
Hi Ziirish.
I've tried with the option v3.3 not successfuly. I've also forced the closing of smart life app on the phone used to retrieve the local key. The hardware is a Teckin sp21.
Thanks

ziirish, 2020-07-15:
You may need to enable debugging and have a look at the logfile as explained here: https://github.com/ziirish/OctoPrint-TuyaSmartplug/issues/5#issuecomment-528269185

capillonpub, 2020-07-20:
hi here is the log (changed ip, id & key):

[2020-07-19 20:40:56,582] DEBUG: Received M80 command, attempting power on of 192.168.0.110.
[2020-07-19 20:40:56,620] DEBUG: {'autoConnect': True, 'v33': True, 'ip': '192.168.0.110', 'btnColor': '#808080', 'autoDisconnectDelay': '2', 'sysCmdOff': False, 'gcodeEnabled': True, 'gcodeOnDelay': '7', 'id': 'ebaxxxxxxxxxxxxx80wrhs', 'slot': 1, 'countdownOnDelay': 0, 'label': 'q5', 'warnPrinting': False, 'gcodeOffDelay': '2', 'sysCmdOnDelay': 0, 'localKey': '7ceb8bxxxxxxxxxx', 'countdownOffDelay': 0, 'sysRunCmdOff': '', 'useCountdownRules': False, 'autoDisconnect': True, 'sysRunCmdOn': '', 'icon': 'icon-bolt', 'displayWarning': True, 'autoConnectDelay': '7', 'currentState': 'unknown', 'sysCmdOn': False, 'sysCmdOffDelay': 0}
[2020-07-19 20:41:03,626] DEBUG: Turning on q5.
[2020-07-19 20:41:03,626] DEBUG: Sending command: info to q5
[2020-07-19 20:41:03,903] DEBUG: Something went wrong while running the command
[2020-07-19 20:41:03,905] DEBUG: {'autoConnect': True, 'v33': True, 'ip': '192.168.0.110', 'btnColor': '#808080', 'autoDisconnectDelay': '2', 'sysCmdOff': False, 'gcodeEnabled': True, 'gcodeOnDelay': '7', 'id': 'ebaxxxxxxxxxxxxx80wrhs', 'slot': 1, 'countdownOnDelay': 0, 'label': 'q5', 'warnPrinting': False, 'gcodeOffDelay': '2', 'sysCmdOnDelay': 0, 'localKey': '7ceb8bxxxxxxxxxx', 'countdownOffDelay': 0, 'sysRunCmdOff': '', 'useCountdownRules': False, 'autoDisconnect': True, 'sysRunCmdOn': '', 'icon': 'icon-bolt', 'displayWarning': True, 'autoConnectDelay': '7', 'currentState': 'unknown', 'sysCmdOn': False, 'sysCmdOffDelay': 0}
[2020-07-19 20:41:03,905] DEBUG: Sending command: on to q5
[2020-07-19 20:41:04,505] DEBUG: Something went wrong while running the command
[2020-07-20 07:19:00,985] DEBUG: Checking status of q5.
......
[plugin_tuyasmartplug_debug.log](https://github.com/ziirish/OctoPrint-TuyaSmartplug/files/4947320/plugin_tuyasmartplug_debug.log)

Don't know how to interpret this!
thanks

ziirish, 2020-07-20:
Can you try again with v3.3 off?

capillonpub, 2020-07-20:
Sure!

disabled v3.3, rebooted printer, octopi and unpowered plug. then:
[2020-07-20 14:01:02,384] DEBUG: Checking status of q5.
[2020-07-20 14:01:02,386] DEBUG: Sending command: info to q5
[2020-07-20 14:01:05,499] DEBUG: Network error
[2020-07-20 14:02:43,807] DEBUG: Checking status of q5.
[2020-07-20 14:02:43,809] DEBUG: Sending command: info to q5
[2020-07-20 14:02:46,921] DEBUG: Network error
[2020-07-20 14:02:52,922] DEBUG: Checking status of q5.
[2020-07-20 14:02:52,922] DEBUG: Sending command: info to q5
[2020-07-20 14:03:07,387] DEBUG: Network error
[2020-07-20 14:03:12,716] DEBUG: Checking status of q5.
[2020-07-20 14:03:12,717] DEBUG: Sending command: info to q5
[2020-07-20 14:03:15,867] DEBUG: Network error
[2020-07-20 14:07:08,237] DEBUG: Checking status of q5.
[2020-07-20 14:07:08,239] DEBUG: Sending command: info to q5
[2020-07-20 14:07:11,347] DEBUG: Network error
[2020-07-20 14:07:13,912] DEBUG: Checking status of q5.
[2020-07-20 14:07:13,912] DEBUG: Sending command: info to q5
[2020-07-20 14:07:14,467] DEBUG: Network error

kinstray, 2020-10-19:
I found this method the easiest

Follow this guide first: https://developer.tuya.com/en/docs/iot/open-api/quick-start/quick-start1?id=K95ztz9u9t89n

Make sure you import your devices using 'Link devices by App account'.  Also under 'API Group', enable the API's giving any random reason.  I'm not sure which, I just enabled them all.  Remember there are 3 pages!

You'll need your Client ID, Client Secret and Device ID for the plug you need the Local ID for.  Those are all in the Tuya [portal](https://iot.tuya.com/cloud/).

Written by @Moudoux on 23 May, thanks to them for this bit! :

1. Install [Postman](https://www.postman.com/downloads/)
2. Download and import the [environment and collection](https://images.tuyacn.com/smart/docs/API_postman_collection.json) in postman
3. Set up the variables according to your [Tuya developer profile](https://iot.tuya.com/cloud/) client_id, device_id (from the app as described in this [post](https://developer.tuya.com/en/docs/iot/open-api/quick-start/quick-start1?id=K95ztz9u9t89n)) and url (region based, i.e. https://openapi.tuyaeu.com for EU) - Make sure you've enabled the API's (mentioned above) and wait 5 to 10 mins or so for them to take effect!
4. Run the access_token task in collections in postman
5. Run the {{url}}/v1.0/devices/{{device_id}} request to get your local_key

Found this much easier than messing around with SSL certificates on Android devices and Proxy apps!

tavdog, 2021-02-03:
Wow, I spent about 5 hours trying to get this to work. I did get the id and key but it still never worked.  Then I realized that Tuya has IFTTT actions enabled and this is a much easier way to get control of the Tuya smart plug for octoprint : https://www.reddit.com/r/3Dprinting/comments/aqb853/easycheap_remote_power_control_with_octoprint/

WimVDK, 2021-02-20:
> Wow, I spent about 5 hours trying to get this to work. I did get the id and key but it still never worked. Then I realized that Tuya has IFTTT actions enabled and this is a much easier way to get [...]

Well i checked your link, i can't even get past step 2 to create the IFTTT apples for my tuya based devices linked in their "Smart Life" app.

michalsovcik, 2021-03-13:
> Wow, I spent about 5 hours trying to get this to work. I did get the id and key but it still never worked. Then I realized that Tuya has IFTTT actions enabled and this is a much easier way to get [...]

@tavdog Can you please explain how did you created IFTTT applets for your plug? It would helpful for me and @WimVDK.

tavdog, 2021-03-13:
Everything you need is in the reddit link.  If that doesn't work for you then I guess don't use IFTTT.

WimVDK, 2021-03-13:
No it isn't. The triggers you're talking about simply don't exist anymore when logged in on that website

ghosilyk, 2021-05-17:
Hey all, I was about to go down this rabbit hole and I'm sorry if this question doesn't belong here, but I'm curious why this plugin doesn't use a different way to connect to Tuya devices. Other things I've setup on the Pi like HomeAssistant and HomeBridge just ask for user/pass and your Tuya devices show up. I noticed a few different libraries to talk to Tuya and figured that this plugin must not be using the same library those other projects are using. I was thinking of starting over with one of those other libraries, but I really don't know what I'm doing so I would spend a lot of time re-figuring out a lot of what's already been figured out here. Anyway, just wondering if there's a plan to switch to another method that doesn't require IP, Device ID and LocalKey, etc.

capillonpub, 2021-05-18:
Hi,
I was not able anymore to find the right key on new tuya devices with all described methods. I found a workaround using Sequematic webhooks that are fully compliant and really easier to integrate.  I cannot give you the tutorial used as it is in french, but here is a sequematic link with informations for Tuya connection (and documentation is clear).
https://sequematic.com/news/2649/tuya-smart-life-integration?utm_source=twitter&utm_medium=post&utm_campaign=tuya-announcement

Deses, 2021-05-19:
Did anybody else's plugin stopped working yesterday? It says that the IP/Hostname of the plug is wrong but it isn't, it's in the same IP as always. I mapped the plug's MAC address to a static IP with Pi-Hole and that didn't change.

I can still turn on and off the plug with Google Home but not within Octoprint...