ziirish/OctoPrint-TuyaSmartplug #9: add region

giloser, 2020-03-24:
Hi, 

after a long search to get the local key I finally get them but I got an network error on the log when I tried to send a command.
To get my localkey I had to use the region option with eu as value.

Could this be the reason of my error?

thanks a lot

ziirish, 2020-03-24:
Hello,

I don't think regions have anything to do with your problem.
This plugin interacts directly with your plug, that's why it needs is local ip address.

Now there are mainly two things that can trigger connection error:

- either there is a running application interacting with your plug (there can be only one connection)
- or you need to use version 3.3 of the protocol
- (or both)

giloser, 2020-03-24:
Hi thanks for the quick answer,

so I cannot have them in the apps like smart life and in octoprint?

thanks

ziirish, 2020-03-24:
Yes you can, but you have to make sure the app is not running when you want octoprint to be able to control it.

closed 2020-07-15 by ziirish