ziirish/OctoPrint-TuyaSmartplug #10: Python3?

faz, 2020-05-04:
Hi, is it easy to update this plugin for python3 to use with the new Octoprint version? Not really a python person, sorry....

ziirish, 2020-05-16:
Interesting, I'll have a look when I have a chance.

0xAalaoui, 2020-06-08:
Simply add  `__plugin_pythoncompat__ = ">=2.7,<4"` to `__init__.py`

Source: [https://docs.octoprint.org/en/master/plugins/python3_migration.html#telling-octoprint-your-plugin-is-python-3-ready](https://docs.octoprint.org/en/master/plugins/python3_migration.html#telling-octoprint-your-plugin-is-python-3-ready)

faz, 2020-06-09:
> Simply add `__plugin_pythoncompat__ = ">=2.7,<4"` to `__init__.py`
> 
> Source: https://docs.octoprint.org/en/master/plugins/python3_migration.html#telling-octoprint-your-plugin-is-python-3-ready

Thanks I'll give this a shot!

commit 2e0b0d18 (2020-10-16): chore: python3 compatibility (fixes #10 and #19)
closed 2020-10-16 by ziirish