ziirish/OctoPrint-TuyaSmartplug #11: pyaesImportError: No module named pyaes

FrancYescO, 2020-05-11:
Cannot install from plugin manager

```
Installing plugin "OctoPrint-TuyaSmartplug" from https://github.com/ziirish/OctoPrint-TuyaSmartplug/archive/master.zip.../home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install https://github.com/ziirish/OctoPrint-TuyaSmartplug/archive/master.zip --no-cache-dirLooking in indexes: https://pypi.org/simple, https://www.piwheels.org/simpleCollecting https://github.com/ziirish/OctoPrint-TuyaSmartplug/archive/master.zipDownloading https://github.com/ziirish/OctoPrint-TuyaSmartplug/archive/master.zipRequirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-TuyaSmartplug==0.2.2) (1.4.0)Collecting pyaes==1.6.1 (from OctoPrint-TuyaSmartplug==0.2.2)Downloading https://files.pythonhosted.org/packages/44/66/2c17bae31c906613795711fc78045c285048168919ace2220daa372c7d72/pyaes-1.6.1.tar.gzCollecting pytuya==7.0.5 (from OctoPrint-TuyaSmartplug==0.2.2)Downloading https://files.pythonhosted.org/packages/4a/69/b20175759e97294f694939ab9bc4be378b3c60ef2d1601d80fc3cc6d2daa/pytuya-7.0.5.zipCommand "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-aimEBj/pytuya/Complete output from command python setup.py egg_info:Traceback (most recent call last):File "<string>", line 1, in <module>File "/tmp/pip-install-aimEBj/pytuya/setup.py", line 9, in <module>import pytuyaFile "pytuya/__init__.py", line 28, in <module>import pyaes  # https://github.com/ricmoo/pyaesImportError: No module named pyaes----------------------------------------Error!Could not parse output from pip, see plugin_pluginmanager_console.log for generated output
```

Bingspot205, 2020-05-16:
hi, did you solve your problem ? i have this problem too

FrancYescO, 2020-05-16:
Nope..

Bingspot205, 2020-05-16:
it's good for me... 
you have to connect to your raspberry pi via putty software and enter the following command line:

source ~ / oprint / bin / activate; pip install pyaes == 1.6.1

ziirish, 2020-05-16:
Hello,

It's a known issue already reported in #6.
This error is due to one of the plugin dependency which didn't handle its own dependencies properly.
I contributed a fix upstream and was waiting for a new release since then.

This release finally happened a couple days ago so I will publish a new version of this plugin which should hopefully address this issue for good.

ziirish, 2020-05-16:
I have now published a new version.

vv4yno, 2020-05-18:
Im still getting that same error as well.
Also,
source ~ / oprint / bin / activate; pip install pyaes == 1.6.1

gets me the following error - I have NFI what it means though :)

Invalid requirement: '=='
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/pip/_internal/req/constructors.py", line 253, in install_req_from_line
    req = Requirement(req)
  File "/usr/share/python-wheels/packaging-19.0-py2.py3-none-any.whl/packaging/requirements.py", line 97, in __init__
    requirement_string[e.loc : e.loc + 8], e.msg
InvalidRequirement: Parse error at "'=='": Expected W:(abcd...)

ziirish, 2020-05-19:
You have to remove whitespaces around `==` => `pip install pyaes==1.6.1`

Also I confirm there is still a problem upstream so I am re-opening this issue.

Archmonger, 2020-06-02:
Am hopeful for a fix!

Bellumatt, 2020-06-06:
Still not fixed .... really sorry for that.
Looking forward for a solution.
What if pyaes is installed before the plugin directly on the pi??

Pitreaper, 2020-06-21:
I am experiencing the same issue trying to install the plugin

SpaceDiceMan, 2020-06-24:
Same problem here

quloquloqulo, 2020-07-03:
connect via ssh and run the command `source oprint/bin/activate; pip install pyaes==1.6.1
`

ziirish, 2020-10-16:
now fixed in 0.3.0

closed 2020-10-16 by ziirish