ziirish/OctoPrint-TuyaSmartplug #12: Cant install due to error

ravenaus, 2020-06-07:
Hi

here is the current error running on octoPrint v1.4.0

Downloading https://files.pythonhosted.org/packages/44/66/2c17bae31c906613795711fc78045c285048168919ace2220daa372c7d72/pyaes-1.6.1.tar.gz
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='www.piwheels.org', port=443): Read timed out. (read timeout=15)",)': /simple/pytuya/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ReadTimeoutError("HTTPSConnectionPool(host='www.piwheels.org', port=443): Read timed out. (read timeout=15)",)': /simple/pytuya/
Collecting pytuya==7.0.6
Downloading https://files.pythonhosted.org/packages/35/82/a10d977c4a0879c37275e2793d983a8b7ac0f6738e937cdc159d4f3a15c1/pytuya-7.0.6.zip
ERROR: Command errored out with exit status 1:
command: /home/pi/oprint/bin/python2 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-szsUxS/pytuya/setup.py'"'"'; __file__='"'"'/tmp/pip-install-szsUxS/pytuya/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-install-szsUxS/pytuya/pip-egg-info
cwd: /tmp/pip-install-szsUxS/pytuya/
Complete output (7 lines):
Traceback (most recent call last):
File "<string>", line 1, in <module>
File "/tmp/pip-install-szsUxS/pytuya/setup.py", line 9, in <module>
from pytuya.const import __author__, __version__
File "pytuya/__init__.py", line 28, in <module>
import pyaes  # https://github.com/ricmoo/pyaes
ImportError: No module named pyaes
----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

Pitreaper, 2020-06-21:
I am having the same problem trying to install this

SpaceDiceMan, 2020-06-24:
Same problem

ziirish, 2020-07-15:
closing as this is clearly a duplicate of #6

closed 2020-07-15 by ziirish