ziirish/OctoPrint-TuyaSmartplug #13: packed_libs.js  POST http://octopi/api/plugin/tuyasmartplug 500 (INTERNAL SERVER ERROR)

seb49, 2020-06-15:
Hello,

Because I don't understant why tuyasmartplug won't work, I looked in Firefox log and there is in error in  packed_libs.js line 9601

// Do send the request (this may raise an exception)
xhr.send( options.hasContent && options.data || null );

Any idea ?

ziirish, 2020-07-15:
It looks like your smartplug is not returning the expected payload although it is responding.
Are you sure it is a tuya-like hardware?

seb49, 2020-07-17:
Hello,

There is the blob : https://www.amazon.fr/gp/product/B07W3NQ518/ref=ppx_yo_dt_b_asin_title_o07_s00?ie=UTF8&psc=1

It works fine with Jeedom. I can get token and device id with Smart Life app and Packet Capture, it seems to be tuya-like hardware

Regards

Taylor-Carnell, 2021-06-10:
Same problem here with unexpected payload in the logs. Plug works fine in the Tuya Smart app and was able to get key/ID from the API service. Took an age to get due to a recent change with the API, would be good to get this working. 
I went with this instead of the Tasmota route to save time/effort but am starting to suspect it would be easier.