ziirish/OctoPrint-TuyaSmartplug #14: Create a script for integration in OctoPrint-PSUControl

quloquloqulo, 2020-07-01:
It would be nice to be able to integrate the two plugins as it was done for TPLink Smart Plug.

https://github.com/kantlivelong/OctoPrint-PSUControl/wiki/Scripts#tplink-smart-plug
https://raw.githubusercontent.com/kantlivelong/OctoPrint-PSUControl/tplink/extra/scripts/tplink_smartplug/tplink-smartplug_wrapper.sh

a system command that reads the state of the plug would probably be enough

bvrielink, 2020-12-02:
Hi. I needed this as well so I created a script to do exactly this. You can find it here: https://github.com/bvrielink/tuyasmartplug.sh. Hopefully you can use it as well. Let me know if it works for you.

quloquloqulo, 2020-12-04:
Thank you very much! It will be very useful to me.  any plan to add a sensing method in future?

bvrielink, 2020-12-05:
I don't really use it but I added a status option (=sensing) just for you. 😉 Just look at the above url and download version 0.2. To make the status option work you need to enable debug logging in the Tuya Smartplug plugin.

quloquloqulo, 2020-12-05:
Thank you very much indeed! your help is very valuable!
I saw the update you posted but it reports this error to me:
` line 48: DEBUG: Status:: command not found`
I also have the doubt that it might not work properly in case of multiple smart sockets (I have printer power supply, light and 2 relays)
How can I help solve these problems?

bvrielink, 2020-12-05:
I got 3 Tuya Smartplugs here without problem. Maybe grep or tail are not installed? Try "sudo apt-get install coreutils grep" if you're on an Debian/Ubuntu/OctoPi like system. If it does not work create an issue on my Github page so we can debug it there.

Beento, 2022-01-20:
Hi. I was trying to use the plugin but wasn't able to install it. I can't find it on the octoprint plugins or add it through link or file