ziirish/OctoPrint-TuyaSmartplug #15: ERROR: Command errored out with exit status 1: python setup.py egg_info while installing "octoprint-tuyasmartplug"

definitely-not-a-t-rex, 2020-07-02:
I'm unable to install the plugin because while trying to install it I get:

```
Installing plugin "OctoPrint-TuyaSmartplug" from https://github.com/ziirish/OctoPrint-TuyaSmartplug/archive/master.zip...
/home/pi/OctoPrint/venv/bin/python2 -m pip --disable-pip-version-check install https://github.com/ziirish/OctoPrint-TuyaSmartplug/archive/master.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/ziirish/OctoPrint-TuyaSmartplug/archive/master.zip
Downloading https://github.com/ziirish/OctoPrint-TuyaSmartplug/archive/master.zip
Requirement already satisfied: OctoPrint in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint-TuyaSmartplug==0.2.3) (1.4.0)
Collecting pyaes==1.6.1
Downloading pyaes-1.6.1.tar.gz (28 kB)
Collecting pytuya==7.0.6
Downloading pytuya-7.0.6.zip (11 kB)
ERROR: Command errored out with exit status 1:
command: /home/pi/OctoPrint/venv/bin/python2 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-3GpkLM/pytuya/setup.py'"'"'; __file__='"'"'/tmp/pip-install-3GpkLM/pytuya/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-install-3GpkLM/pytuya/pip-egg-info
cwd: /tmp/pip-install-3GpkLM/pytuya/
Complete output (7 lines):
Traceback (most recent call last):
File "<string>", line 1, in <module>
File "/tmp/pip-install-3GpkLM/pytuya/setup.py", line 9, in <module>
from pytuya.const import __author__, __version__
File "pytuya/__init__.py", line 28, in <module>
import pyaes  # https://github.com/ricmoo/pyaes
ImportError: No module named pyaes
----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output
```
The problem seem to be with pyaes, has anybody who got the issue managed to fix it?

quloquloqulo, 2020-07-03:
connect via ssh and run the command `source oprint/bin/activate; pip install pyaes==1.6.1`

Nathan116, 2020-07-12:
> connect via ssh and run the command `source ~/oprint/bin/activate; pip install pyaes==1.6.1`

So. I'm running into the same issue, but I'm not running Octoprint from a Pi, I'm running if from a windows PC (older thing I had sitting around), so I don't think I can connect via SSH. How would I go about fixing the issue?

ziirish, 2020-07-15:
The Python doc says `C:\> <venv>\Scripts\activate.bat` [here](https://docs.python.org/3/library/venv.html)
Where `<venv>` is the path where octoprint should be installed on your system.
Once you have activated your venv, you can run the ` pip install pyaes==1.6.1` command.

Nathan116, 2020-07-15:
Thank you so much, it worked great! Now, to figure out how to get the local key (I'm not great at computer coding so all this is quite a project for me)

ziirish, 2020-07-15:
You may find some tips & tricks in #8

ziirish, 2020-10-16:
now fixed in 0.3.0

closed 2020-10-16 by ziirish