ziirish/OctoPrint-TuyaSmartplug #16: status unknown for <ip of smart plug>

hanzala-123, 2020-07-23:
Hi, i am new to octoprint and wanted to set everything up before my printer arrives (Ender 3 v2) and i have installed the plugin, obtained all the device info using the tuya api and the app and have entered everything correctly into the plugin settings. when i click on the switch (greyed out ) on the navbar, it returns the message:

Status unknown for 192.168.1.80. Double check IP Address\Hostname in tuyasmartplug Settings.

[image]

[image]

[image]

surenz, 2020-07-28:
+1 the issue and I found that if I disable the v3.3 protocol checkbox the smartplug is discoverable but it does not execute command for shut down (no error in log neither on the UI). Of course my device has firmware 1.0.5 which means I have to use v.3.3. checkbox.

Chuckame, 2020-09-09:
Same issue !

TempestWales, 2020-09-11:
Same issue, My power Icon just spins.

war1000, 2020-09-18:
Same issue for me. checking the v3.3 says unknown status and unchecking spins the bolt icon.

[2020-09-18 02:15:16,616] DEBUG: Checking status of ZZPlug7.
[2020-09-18 02:15:16,617] DEBUG: Sending command: info to ZZPlug7
[2020-09-18 02:16:05,623] DEBUG: Checking status of ZZPlug7.
[2020-09-18 02:16:05,624] DEBUG: Sending command: info to ZZPlug7

csabal, 2020-10-04:
eh, exact the same issue here. :( I was so happy to find this plugin. Managed to get the ID, localkey etc, from Tuya API, and now it is for nothing. :( Could you pls check what is gone wring w this great plugin?

Deses, 2020-10-11:
Weird! I have protocol 3.3 enabled and it works perfectly.
[image]

Are you absolutely sure that the smart plug is using that IP? I manually assigned the IP of the plug using the MAC address and the DHCP options of PiHole (or your router should have the same thing).

TempestWales, 2020-10-15:
> Weird! I have protocol 3.3 enabled and it works perfectly.
> 
> Are you absolutely sure that the smart plug is using that IP? I manually assigned the IP of the plug using the MAC address and the DHCP options of PiHole (or your router should have the same thing).
> 
> [image]

Yes. Either static or DHCP, I can ping the plug from my PC or the Raspberry Pi.

Sorcier, 2020-11-11:
Hi,

I have the same problem, plug work correctly in the app, IP is correct, i found local key with Tuya API.
But impossible to control my plug from the plugin i have an error : 
Tuya Smartplug Error
Status unknown for 192.168.1.22. Double check IP Address\Hostname in tuyasmartplug Settings.

In the log we have that, it's light : 
[2020-11-11 18:47:56,846] DEBUG: Checking status of Prusa i3 MK3S.
[2020-11-11 18:47:56,846] DEBUG: Sending command: info to Prusa i3 MK3S
[2020-11-11 18:47:56,930] DEBUG: Something went wrong while running the command

[image]

[image]

FrancYescO, 2020-11-23:
Same here, firmware of plug 1.1.4
Version 3.3 checkbox does not help.

TempestWales, 2020-11-26:
I've checked the Firmware version of my Teckin SP23/27 plugs, and they are 1.0.3. and 1.0.0 respectively. Not sure if this makes a difference or incompatible with the Plugin.

moogster, 2021-01-07:
Hi has anyone got Teckin SP23 plugs to work successfully yet?!  I have tried everything.  If I enable the 3.3 check box I get failure to get the status of the plug. Without it enabled I get Sending command: info to ANET_A8 but nothing else happens.
Thanks

devjulie, 2021-02-15:
I've got the same issue, checking the 3.3 checkbox makes no difference, I get the same error about the IP address but its definitely correct. I have setup the plug with "Homebridge Tuya Lan" plugin using the same id and key with no issues whatsoever so must be an issue with this plugin.

jtong77, 2021-03-27:
I had the same issues. After searching all over, I found this thread: https://github.com/ziirish/OctoPrint-TuyaSmartplug/issues/5

> "What I know, is those plugs only support ONE client at the time to interact with them.
> Could you make sure your phone/tablet/whatever has the app installed are turned off (just switch the wifi off) and then try again?"

So I put my phone in airplane mode and after a minute or so it works.

FrancYescO, 2021-03-28:
so a solution can be to share the plug to a new tuya account? i have the same plug on home assistant (and phone, all using same account), using it from octoprint with this plugin remain totally unreliable.

speedysk1, 2021-05-24:
Same issue here.  It used to work with 3.3 unchecked, then it just stopped working.  Maybe a firmware update on the plug?  In any case it doesn't work anymore.

Deses, 2021-05-25:
> Same issue here. It used to work with 3.3 unchecked, then it just stopped working. Maybe a firmware update on the plug? In any case it doesn't work anymore.

Yeah, my plug also stopped working through Octoprint. This fucking sucks.

ziirish, 2021-05-25:
Hello,

Sorry to hear that. Like I said before, I am no longer using those plugs... However, I will try to see how home-assistant handles them and try to rework the plugin accordingly.

In the meantime, if anyone is willing to help you are more than welcome :)

speedysk1, 2021-05-30:
> Hello,
> 
> Sorry to hear that. Like I said before, I am no longer using those plugs... However, I will try to see how home-assistant handles them and try to rework the plugin accordingly.
> 
> In the meantime, if anyone is willing to help you are more than welcome :)

Thanks for the update.  If there is anything I can do to help debug, I'll be happy to.  I don't do enough coding to want to try to debug the code myself however.  Sorry.

ziirish, 2021-06-15:
Hello,

I have just pushed new code using the same kind of API as what home-assistant uses.
The drawback of this is that octoprint MUST be able to connect to the Internet (in order to talk to the Tuya cloud API). But on the other hand, this allows you to keep using your mobile app to control your plugs and removes the burden of finding the local key and secret.

Like I said, I cannot test it as I don't have such plugs anymore so for now everything I did was based on what I could gather on the Internet.

Note that if you intend to use this new version of my plugin, your settings will be reset because I no longer need the local key and secret, and because instead I need your Tuya username, password and country code.

I have not made a proper release yet so I guess you will need to download this repo as a zip file, then install the plugin from the zip archive within octoprint.

Feedback more than welcome!

ziirish, 2021-06-15:
Oh, one little more thing: when configuring the plugin, I need a plug `label`. This must be the name you gave to the plug within the Tuya application.

ziirish, 2021-06-16:
I have had a deeper look and I'm not sure this lib is better for this plugin.
From what I can see, we don't have all the features we could expect. For instance, we cannot setup a timer.
Also, although I don't have any device to play with anymore, I still have an account on Tuya Cloud so I tried a couple API calls, but it *seems* the API is rate limited because I sometime see messages like: `{'msg': 'you can query once in 120 seconds', 'code': 'FrequentlyInvoke', 'payloadVersion': 1}`

kikirikiki, 2023-02-08:
Hi, thanks a lot for your work on this plugin. I have the same issue or at least the flash turn red on the octoprint interface but the switch do not close. What could be the reason? You mention there was a zip file with new update, where can I found that?

If you are not using this plugin anymore, which one do you recommend to use with a wifi switch?

Thanks a lot!

xia0, 2023-07-24:
I found a workaround by accessing the api directly using python and the [PSU Control](https://plugins.octoprint.org/plugins/psucontrol/) plugin. If you've gone through the trouble of setting up a developer account, then you should already have the required credentials.

You will need to install [tuya-connector-python](https://github.com/tuya/tuya-connector-python). See [this guide](https://developer.tuya.com/en/demo/python-iot-development-practice) if you want to know more.

Here's my code:
```python
# https://github.com/tuya/tuya-connector-python
# https://developer.tuya.com/en/demo/python-iot-development-practice

import sys
import logging
from tuya_connector import TuyaOpenAPI, TUYA_LOGGER

if (len(sys.argv) > 1):

  ACCESS_ID = "xxx"
  ACCESS_KEY = "xxx"
  API_ENDPOINT = "https://openapi.tuyaeu.com"
  DEVICE_ID = "xxx"

  # Enable debug log
  #TUYA_LOGGER.setLevel(logging.DEBUG)

  # Init OpenAPI and connect
  openapi = TuyaOpenAPI(API_ENDPOINT, ACCESS_ID, ACCESS_KEY)
  openapi.connect()

  # get status of switch
  if sys.argv[1] == "status":
    # Get the status of a single device
    response = openapi.get("/v1.0/iot-03/devices/{}/status".format(DEVICE_ID))
    for r in response['result']:
      if r['code'] == "switch_1":
        print(r['value'])
        if r['value'] == False:
          sys.exit(1)
        else:
          sys.exit(0)

  else:
    # set status
    state = None
    if (sys.argv[1] == "on" or sys.argv[1] == "1"):
      state = True
    elif (sys.argv[1] == "off" or sys.argv[1] == "0"):
      state = False

    if state != None:
      commands = {'commands': [{'code': 'switch_1', 'value': state}]}
      openapi.post('/v1.0/iot-03/devices/{}/commands'.format(DEVICE_ID), commands)
      sys.exit()

print("commands: status, on, off")
sys.exit(-1)
```

Then use the [PSU Control](https://plugins.octoprint.org/plugins/psucontrol/) plugin to control the switching. Set both switching and sending methods to 'System Command' and run the python script with the arguments either `on`, `off` or `status`.

danivalls, 2023-08-18:
Sorry for bumping an old issue but I'm still having this problem. Hasn't it been solved?

Thaer-Jleilati, 2024-09-14:
same, still having problem