ziirish/OctoPrint-TuyaSmartplug #17: useCountdownRules doesn't work

yuri83git, 2020-08-06:
if i insert a timer (ex 30s) the plug doesn't shut down.

To fix the bug, i had to modify __init___.py:

line 133:
    chk = self.sendCommand('countdown', plug['label'], **int**(plug['countdownOnDelay']))
line 157:
    chk = self.sendCommand('countdown', plug['label'], **int**(plug['countdownOffDelay']))

yuri83git, 2020-08-07:
here some debug before my mod, look the print "arg 6" -> 6 is my value for countdown timer:
[2020-08-06 10:25:54,391] DEBUG: Sending command: countdown to plug, args  6
[2020-08-06 10:25:55,001] DEBUG: Status: {u'devId': u'xxxx', u'dps': {u'1': True, u'2': 0}}

and this is after my mod, 32 is my new time:
[2020-08-06 10:43:28,611] DEBUG: Sending command: countdown to plug, args  32
[2020-08-06 10:43:29,204] DEBUG: Status: {u'devId': u'01200474600194928bf1', u'dps': {u'1': True, u'2': 32}}

ziirish, 2020-10-10:
Hello,

Thanks for the report and patch.
I will include it in the next release.

Deses, 2020-10-11:
I can confirm this fix working.

commit 14573130 (2020-10-16): fix: countdown didn't work due to wrong argument type (fixes #17) Huge thanks to @yuri83git for the patch
closed 2020-10-16 by ziirish