ziirish/OctoPrint-TuyaSmartplug #19: [Feature Request] Python 3 Compatibility

quloquloqulo, 2020-09-11:
Please add Python 3 Compatibility as outlined here.

Deses, 2020-10-09:
I agree! This plugin is an integral part of my setup, I can't upgrade Octoprint until this one gets updated.

ziirish, 2020-10-10:
Hello,

I'm on it, but I would need beta testers since I'm not using any tuya smart plug anymore and I also don't have (yet) a python3 setup for octoprint.

Any vonlunteer?

bepstein111, 2020-10-10:
Absolutely count me in!

ziirish, 2020-10-10:
Anyone wanting to give it a shot, here is the upcoming 0.3.0 build:

[OctoPrint-TuyaSmartplug.tar.gz](https://github.com/ziirish/OctoPrint-TuyaSmartplug/files/5360011/OctoPrint-TuyaSmartplug.tar.gz)

Deses, 2020-10-11:
Does it work with Python 2 installations?

Deses, 2020-10-11:
[OctoPrint-TuyaSmartplug.zip](https://github.com/ziirish/OctoPrint-TuyaSmartplug/files/5362296/OctoPrint-TuyaSmartplug.zip)
Here's the same file as @ziirish posted but in zip form to manually install through OctoPrint.

ziirish, 2020-10-11:
It *should*, although I didn't have a chance to give it a proper try yet.

Also the pytuya code is now embedded to avoid the installation issues on python2 setup.

Deses, 2020-10-11:
Something is wrong... I installed it but the plugin is nowhere to be found in the web interface. Not even in the disabled plugin list.

[octoprint.log](https://github.com/ziirish/OctoPrint-TuyaSmartplug/files/5362308/octoprint.log)

Here's what I found in the logs:
```2020-10-12 00:53:24,548 - octoprint.plugin.core - ERROR - Error loading plugin tuyasmartplug
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 972, in _import_plugin
    module = _load_module(module_name, spec)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 71, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tuyasmartplug/__init__.py", line 14, in <module>
    from octoprint_tuyasmartplug.utils import pytuya
ImportError: No module named utils```

And this is the install log: 

`Installing plugin from uploaded archive...
/home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install file:///tmp/tmp37v4gW.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /tmp/tmp37v4gW.zip
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-TuyaSmartplug==0.3.0) (1.4.2)
Requirement already satisfied: pyaes==1.6.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-TuyaSmartplug==0.3.0) (1.6.1)
Requirement already satisfied: pycrypto==2.6.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-TuyaSmartplug==0.3.0) (2.6.1)
Requirement already satisfied: futures<4,>=3.3; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TuyaSmartplug==0.3.0) (3.3.0)
Requirement already satisfied: semantic-version<3,>=2.8.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TuyaSmartplug==0.3.0) (2.8.5)
Requirement already satisfied: future<1,>=0.18.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TuyaSmartplug==0.3.0) (0.18.2)
Requirement already satisfied: blinker<2,>=1.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TuyaSmartplug==0.3.0) (1.4)
Requirement already satisfied: Flask-Assets<3,>=2.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TuyaSmartplug==0.3.0) (2.0)
Requirement already satisfied: unidecode<0.05,>=0.04.14 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TuyaSmartplug==0.3.0) (0.4.21)
Requirement already satisfied: PyYAML<6,>=5.3.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TuyaSmartplug==0.3.0) (5.3.1)
Requirement already satisfied: pylru<2,>=1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TuyaSmartplug==0.3.0) (1.2.0)
Requirement already satisfied: OctoPrint-FileCheck>=2020.07.06 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TuyaSmartplug==0.3.0) (2020.8.7)
Requirement already satisfied: pkginfo<2,>=1.5.0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TuyaSmartplug==0.3.0) (1.5.0.1)
Requirement already satisfied: wrapt<2,>=1.12.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TuyaSmartplug==0.3.0) (1.12.1)
Requirement already satisfied: chainmap<2,>=1.0.3; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TuyaSmartplug==0.3.0) (1.0.3)
Requirement already satisfied: Click<8,>=7.1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TuyaSmartplug==0.3.0) (7.1.2)
Requirement already satisfied: markdown<3.2,>=3.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TuyaSmartplug==0.3.0) (3.1.1)
Requirement already satisfied: watchdog<1,>=0.10.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TuyaSmartplug==0.3.0) (0.10.3)
Requirement already satisfied: netifaces<1,>=0.10.9 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TuyaSmartplug==0.3.0) (0.10.9)
Requirement already satisfied: scandir<2,>=1.10; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TuyaSmartplug==0.3.0) (1.10.0)
Requirement already satisfied: filetype<2,>=1.0.7 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TuyaSmartplug==0.3.0) (1.0.7)
Requirement already satisfied: itsdangerous<2,>=1.1.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TuyaSmartplug==0.3.0) (1.1.0)
Requirement already satisfied: tornado==5.1.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TuyaSmartplug==0.3.0) (5.1.1)
Requirement already satisfied: cachelib<1,>=0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TuyaSmartplug==0.3.0) (0.1)
Requirement already satisfied: werkzeug<2,>=1.0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TuyaSmartplug==0.3.0) (1.0.1)
Requirement already satisfied: typing<4,>=3.7.4.1; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TuyaSmartplug==0.3.0) (3.7.4.1)
Requirement already satisfied: regex!=2018.11.6 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-TuyaSmartplug==0.3.0) (2019.8.19)
[... cut ...]
Stored in directory: /tmp/pip-ephem-wheel-cache-VQZ7JX/wheels/6f/b9/ca/0fd0976269ad0efdf01a638b98c8068c6a21be5b1d332bce3b
Successfully built OctoPrint-TuyaSmartplug
Installing collected packages: OctoPrint-TuyaSmartplug
Attempting uninstall: OctoPrint-TuyaSmartplug
Found existing installation: OctoPrint-TuyaSmartplug 0.3.0
Uninstalling OctoPrint-TuyaSmartplug-0.3.0:
Successfully uninstalled OctoPrint-TuyaSmartplug-0.3.0
Successfully installed OctoPrint-TuyaSmartplug-0.3.0
Done!`

![image](https://user-images.githubusercontent.com/1654959/95692352-74554280-0c25-11eb-917a-0f8b677120b6.png)

Deses, 2020-10-11:
I changed line 14 in `__init__.py` from `from octoprint_tuyasmartplug.utils import pytuya` to `import pytuya` and the installation worked.

[OctoPrint-TuyaSmartplug.zip](https://github.com/ziirish/OctoPrint-TuyaSmartplug/files/5362323/OctoPrint-TuyaSmartplug.zip)

Here's the zip, if anyone wants it.

ziirish, 2020-10-12:
Thanks for trying this version.
Something is probably missing from the manifest, that would explain the lack of the 'utils' module.
Your fix on the other hand is using the upstream version of pytuya which was already present in your venv.

I'll have another look then.
Thanks again for your valuable feedback.

Deses, 2020-10-13:
I tried reverting back my change to `__init__.py` and adding the `utils` folder in the manifest but it didn't work.

```
include README.md
recursive-include octoprint_tplinksmartplug/templates *
recursive-include octoprint_tplinksmartplug/translations *
recursive-include octoprint_tplinksmartplug/static *
recursive-include octoprint_tplinksmartplug/utils *
```

>Your fix on the other hand is using the upstream version of pytuya which was already present in your venv.
I guess I got lucky! :D

ziirish, 2020-10-13:
Ok, so I have made a new build.

You may need to uninstall the old version first though.

[OctoPrint-TuyaSmartplug-0.3.0.zip](https://github.com/ziirish/OctoPrint-TuyaSmartplug/files/5374200/OctoPrint-TuyaSmartplug-0.3.0.zip)

Deses, 2020-10-13:
Perfect! It installed without any issue!

ziirish, 2020-10-16:
Thanks for the tests.

0.3.0 is now released with support for python 3.

closed 2020-10-16 by ziirish