ziirish/OctoPrint-TuyaSmartplug #20: bug - octopi plugin manager install failure

dixieflatline76, 2020-09-11:
Received following error when installing via octopi plugin manager

Collecting pytuya==7.0.6
Downloading https://files.pythonhosted.org/packages/35/82/a10d977c4a0879c37275e2793d983a8b7ac0f6738e937cdc159d4f3a15c1/pytuya-7.0.6.zip
ERROR: Command errored out with exit status 1:
command: /home/pi/oprint/bin/python2 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-OcLqoA/pytuya/setup.py'"'"'; __file__='"'"'/tmp/pip-install-OcLqoA/pytuya/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-install-OcLqoA/pytuya/pip-egg-info
cwd: /tmp/pip-install-OcLqoA/pytuya/
Complete output (7 lines):
Traceback (most recent call last):
File "<string>", line 1, in <module>
File "/tmp/pip-install-OcLqoA/pytuya/setup.py", line 9, in <module>
from pytuya.const import __author__, __version__
File "pytuya/__init__.py", line 28, in <module>
import pyaes  # https://github.com/ricmoo/pyaes
ImportError: No module named pyaes
----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

Uniseco, 2020-09-22:
Hello,
I have the same error. I suppose that is for the python3, but I don't have any idea how program or resolve it.

dixieflatline76, 2020-09-23:
Damn, I am not a Python guy either, but will take a look at that migration guide and see if i can make it work.

BlaydeRunner08, 2020-10-03:
Depending on your OS and install, you'll need to run a version of these commands: 

`source ~/oprint/bin/activate
pip install pyaes==1.6.1`

commit 894bcd37 (2020-10-13): chore: embed pytuya code to fix installation issues (fixes #20, #15 and #11)
closed 2020-10-16 by ziirish