ziirish/OctoPrint-TuyaSmartplug #33: Can't put in IP

AdamTheLawyer, 2021-06-20:
I installed the plugin, but there's nowhere to put the IP or authentication info in for my plug. It just asks for a username, password, and country code. On the "SmartPlug Editor" page, too ... I can edit the label, but that's it.

What am I missing?

goebish, 2021-06-24:
Same here, there are no IP, Device ID and Local Key fields in the settings page and the main dialog asks for an username & password ?

ziirish, 2021-06-24:
Hello,

Yes, sorry, these changes are part of a current experimentation that rely on the Tuya cloud API instead of having to sniff your network to find the Device ID and Local Key.

I though the Octopi's plugin manager was smart enough to follow tags, but it looks it actually just tracks the master branch of this repository...
I will revert my last changes when I have a moment and start a new branch to run my experiences instead.
In the maintime, you can install the last stable version of this plugin by selecting *install from a link* in Octopi's plugin manager and entering the following URL: https://github.com/ziirish/OctoPrint-TuyaSmartplug/archive/refs/tags/0.3.0.zip

AdamTheLawyer, 2021-06-24:
That worked great; let me put the info in.

Still doesn't want to flip the switch, though. I can turn it on and off with tuya-cli, so I think I've got the settings right. It just doesn't flip the switch ... when I click on the icon it just spins.

Are there logs I can look at to try to see what's going on?

AdamTheLawyer, 2021-06-24:
Hey! A little box that says "Version 3.3". I wonder if I have version 3.3!

(facepalm)

OK, now it works. Thank you!

closed 2021-06-24 by AdamTheLawyer