ziirish/OctoPrint-TuyaSmartplug #35: Unable to toggle plug

ICEconchy, 2021-06-27:
Hello, so I've followed the setup and have tried to turn on the plug. But in logs this appears:
[2021-06-28 00:00:55,181] DEBUG: Checking status of Printer.
[2021-06-28 00:00:55,182] DEBUG: Sending command: info to Printer
[2021-06-28 00:00:55,778] DEBUG: Status: b'&\x0end+\x8d\xef\x13C\x9f__w-\xd1W\xe7\xfc\xa7l\xa5\xc3@d\xdaDy.B\x1a\x94\x85'
[2021-06-28 00:03:23,437] DEBUG: Checking status of Printer.
[2021-06-28 00:03:23,438] DEBUG: Sending command: info to Printer
[2021-06-28 00:03:23,994] DEBUG: Status: b'&\x0end+\x8d\xef\x13C\x9f__w-\xd1W\xe7\xfc\xa7l\xa5\xc3@d\xdaDy.B\x1a\x94\x85'
[2021-06-28 00:23:41,302] DEBUG: Checking status of Printer.
[2021-06-28 00:23:41,302] DEBUG: Sending command: info to Printer
[2021-06-28 00:23:41,869] DEBUG: Status: b'&\x0end+\x8d\xef\x13C\x9f__w-\xd1W\xe7\xfc\xa7l\xa5\xc3@d\xdaDy.B\x1a\x94\x85'

rudamaia, 2021-08-01:
I have the same problem.
`[2021-08-01 09:32:42,212] DEBUG: Sending command: info to cr-10 plug
[2021-08-01 09:32:43,098] DEBUG: Status: b'D\xbel\xf4mg\xf3\xc5Z \x13\x13\xfb\x87\xc4\x8e\x7f\x9atb\xb5\xc5\xc0\xeb\x82\xb16\xc9\xe1\x8e\xc6\xe4'
[2021-08-01 09:34:16,575] DEBUG: Checking status of cr-10 plug.
[2021-08-01 09:34:16,576] DEBUG: Sending command: info to cr-10 plug
[2021-08-01 09:34:17,503] DEBUG: Status: b'D\xbel\xf4mg\xf3\xc5Z \x13\x13\xfb\x87\xc4\x8e\x7f\x9atb\xb5\xc5\xc0\xeb\x82\xb16\xc9\xe1\x8e\xc6\xe4'
[2021-08-01 09:34:57,970] DEBUG: Checking status of cr-10 plug.
[2021-08-01 09:34:57,971] DEBUG: Sending command: info to cr-10 plug
[2021-08-01 09:34:58,875] DEBUG: Status: b'D\xbel\xf4mg\xf3\xc5Z \x13\x13\xfb\x87\xc4\x8e\x7f\x9atb\xb5\xc5\xc0\xeb\x82\xb16\xc9\xe1\x8e\xc6\xe4'
[2021-08-01 09:36:20,881] DEBUG: Checking status of cr-10 plug.
[2021-08-01 09:36:20,882] DEBUG: Sending command: info to cr-10 plug
[2021-08-01 09:36:21,821] DEBUG: Status: b'D\xbel\xf4mg\xf3\xc5Z \x13\x13\xfb\x87\xc4\x8e\x7f\x9atb\xb5\xc5\xc0\xeb\x82\xb16\xc9\xe1\x8e\xc6\xe4'`

rudamaia, 2021-08-01:
Got it working marking the "Version 3.3" checkbox in the plug settings.