ziirish/OctoPrint-TuyaSmartplug #49: Plugin Adoption

andrelucca, 2023-03-26:
Hello @ziirish hope you're doing well.
I was just going through the process of making your plugin work but after many failures I decide to take a look into the source files. I made several adjustments after searching what users were saying in your issues, and for my 3D printing setup I make it work pretty well. Even made a nice Wiki for the plugin with a lot of detailed instructions about the plugin and the Tuya Device ID an Key. Unfortunately it seems that your repository don't see new activity for a couple of months, so a simple Pull Request of this fixes didn't seem to work. 
Would you be interested in going through the adoption process of your plugin? By doing this it will be designated to a new maintainer and the Octoprint plugin repository can notify its users of this changes. 
Thanks in advance for your time and attention

andrelucca, 2023-03-28:
Hello @ziirish , thanks for your response. 
First of all, your plugin is a very good ideia, has a lot of potential, and I'm sure it helped and will help a lot of Octoprint users that has the same needs. I totally undersatand the lack of time for side projects as this one and thats why I contacted you to keep your good work. I will follow the process for plugin adoption that the Octoprint maintainers suggested in the ticket you've linked, I really don't know if its possible to "transfer" a Github repo, so I imagine that they will only change the plugin source to the fork of your repo on the official plugin page (sorry, first time dealing with Octoprint plugins). Anuway I'll follow their instructions and will come back with more info, either add a contributor or "transfer" the plugin as a solution.

jneilliii, 2023-03-29:
It is possible to transfer a GitHub repo to another user, but there can't be a fork or plugin with the same name in your account @andrelucca. Once you've verified that you don't have that in your GitHub @ziirish can go into the settings of the repo and at the bottom of the general section there is a transfer button.

[image]

andrelucca, 2023-04-10:
Thanks @ziirish to add me as a maintainer of your plugin. As soon as I can I'll start to merge my corrections/improvements here. 
@jneilliii thanks for your support, I think it is safe now to close the plugin adoption [Issue](https://github.com/OctoPrint/plugins.octoprint.org/issues/1175) on the octoprint repo, if there's nothing else to be done in the adoption process of course.

closed 2023-04-10 by andrelucca