ziirish/OctoPrint-TuyaSmartplug #51: New version stopped working

beerman-uk, 2023-04-18:
I was running version 0.30 which was working perfectly.  I updated and it gave an error that the ID/key was incorrect even though it turned on the printer when I clicked on the icon.  If I then clicked on the grey icon it turned green.  After a print I had it set to turn off after 5 minutes but this wasn't triggered like it was before the update. 

I have 3 printers that I have tried 0.31 on and none worked properly after the update so I've gone back to 0.30.

ViiFii, 2023-04-25:
Same here... is there any way downgrade version to 0.30?

beerman-uk, 2023-04-25:
> Same here... is there any way downgrade version to 0.30?

If you uninstall the newer version there are 2 options, click the one that retains your data. 
Reboot octoprint
Install from a plugin from a URL and use the following URL: https://github.com/ziirish/OctoPrint-TuyaSmartplug/archive/refs/tags/0.3.0.zip
Reboot octoprint 

It should be back exactly as it was.  Now go into the Software Updates section of the settings and turn off notifications for the TuyaSmartPlug plugin so you don't accidentally update it again.

ViiFii, 2023-04-25:
Thanks, mate!

andrelucca, 2023-05-01:
Hello @beerman-uk sorry for this inconvenience with the new release. Just released the version 0.3.2 trying to fix it, hope it works.

closed 2023-05-01 by andrelucca