ziirish/OctoPrint-TuyaSmartplug #62: Set everything right but not working

Carlsans, 2024-02-08:
I'm getting this bug :
Status unknown for 192.168.2.179. Double check IP Address\Device ID\Local Key in Octoprint-TuyaSmartplug Settings
Everything is set right I'm 100 % sure. Followed the guide and everything.  Have two plugs and failed on both. Have to pi and failed on both. The plugs work fine using the app.

Carlsans, 2024-02-08:
Part of the log with id and localkey deleted.

[2024-02-08 12:08:31,103] DEBUG: Turning on Ender.
[2024-02-08 12:08:31,103] DEBUG: Sending command: info to Ender
[2024-02-08 12:08:31,319] DEBUG: Something went wrong while running the command ord() expected a character, but string of length 0 found
[2024-02-08 12:08:31,323] DEBUG: {'autoConnect': True, 'autoConnectDelay': 10, 'autoDisconnect': True, 'autoDisconnectDelay': 0, 'btnColor': '#808080', 'countdownOffDelay': 0, 'countdownOnDelay': 0, 'currentState': 'off', 'displayWarning': True, 'gcodeEnabled': False, 'gcodeOffDelay': 0, 'gcodeOnDelay': 0, 'icon': 'icon-bolt', 'id': 'dfxfxfdfdsfsdfsdf', 'ip': '192.168.2.178', 'label': 'Ender', 'localKey':ddsffsdfsdfsdfsdffsd, 'slot': 1, 'sysCmdOff': False, 'sysCmdOffDelay': 0, 'sysCmdOn': False, 'sysCmdOnDelay': 0, 'sysRunCmdOff': '', 'sysRunCmdOn': '', 'useCountdownRules': False, 'v33': True, 'warnPrinting': False}
[2024-02-08 12:08:31,323] DEBUG: Sending command: on to Ender
[2024-02-08 12:08:32,341] DEBUG: Something went wrong while running the command ord() expected a character, but string o

Carlsans, 2024-02-08:
I have been able to connect to my plug via python lib tinytuya. [python-tuya] is no longer maintained(This repository has been archived by the owner on Sep 4, 2021). I guess rewiring the code using tinytuya would be an option.

[image]

Carlsans, 2024-02-09:
I've seen that a tinytuya branch was already created but the code was not compatible with 3.4 tuya version. I forked the branch and modifyed the code so that it use 3.4 instead of 3.3. It work. https://github.com/Carlsans/OctoPrint-TuyaSmartplug/tree/tinytuya
You need to have the 3.3 box checked. Would be easy just to add a few radiobox to choose from the newer versions.

kattekatkat, 2024-02-16:
Thank you very much for both to Ziirish for the original version and this updated one from Carlsans. :-) I have been using the plugin for a long time but it also stopped working for me. With the new version from the 15 feb. it is almost working for me again. I'm not able to get the turn off with g-code working anymore. Nothing happens. Also it seems like the status doesn't update after turning the plug on : If I press the icon on the top of the screen the printer turns on, but the icon stays gray. If i click on the icon again is turn green, indicating that the plug is on. Have you managed to get the G-code turn off working ?

Best regards Jesper

Carlsans, 2024-02-16:
My last version has this issue solved. Was an indent problem.
https://github.com/Carlsans/OctoPrint-TuyaSmartplug

kattekatkat, 2024-02-16:
Is this one different than the one from yesterday? I'm using the one from yesterday and having problem with the G-code turn off (I'm using the one with the menu added so that is possible to select Tuya version, im using 3.4). I 've also tried the older version that you linked to in post 3, but the new one works much better.

Carlsans, 2024-02-17:
i deleted the master branch that was wrong and made last corrections.will
evolve in time. my version gcodes end command work.

kattekatkat, 2024-02-17:
I tried again today and it's stille the same. Not able to turn the printer off with G-code M81. I am pretty sure that I was using the latest version yesterday evening. The behavior is the same. Still getting an error "Status unknown for 192.168.2.233. Double check IP Address\Hostname in tuyasmartplug Settings." when turning the plug on and the status (icon color) is not updated right away. (the plug does turn on) I've set "pulling of status" to 1 minut, so it updates within 1 minut, but not directly after I press the icon. The M81 doesn't do anything.

kattekatkat, 2024-02-17:
I don't know how i can uninstall it anymore that I have done so far. Have used the uninstall and clean up data function and restarted before and after uninstalling and installing. Still no luck with M81 but as a workaround i've set up home assistant to turn off the plug for the printer after it's done printing.

Carlsans, 2024-02-19:
I've tested the link I sent you on another pi. I had to delete the data
from the old version so it could work. It's still really buggy but the M81
work. Your better off waiting a little longer until I clear all the bugs.
[image: image.png]

Carlsans, 2024-02-19:
The latest update is a major improvement in stability. Please backup your
info because update will erase it, you will need to write it back.

Carlsans, 2024-02-28:
The latest update of my repo is working quite well. Go take a look at the
video at the top of the readme file.
https://github.com/Carlsans/OctoPrint-TuyaSmartplug

kattekatkat, 2024-02-28:
is  it only possible to add plugs by discovering the with the API/secret method? (and by that adding all the plugs that is connected to the Tuya account?)

Carlsans, 2024-02-28:
You can add them the old way too. You can delete the plugs you dont want.

icedarts88, 2024-04-18:
Hi  @Carlsans 

In the video you say to type M1234 but in the pluggin it says G1234.
Which is the right one?

Carlsans, 2025-01-22:
> Hello @Carlsans <https://github.com/Carlsans>
> I set up your fork of plugin multiple times, I still can't get it working,
> I was hoping your video might help me but it's nowhere to be find on your
> page, any change to reup or making readme?
>
> When I use API key and M1234 method after while I just got empty terminal
> and nothing more, when adding the old way, with ID and local_key extracted
> directly via tinytuya it also don't work. Any help would be appreciated,
> thanks :)

I remembered a bug I had when I tried install the plugin after resetting
the plug. Check that every information using M1234 has been transferred to
the plug form.
Any way, it might be broken because of a tuya update. I really regret using
that much time on such a bullshit platform as tuya. Really hard to set up
and quite undependable.
If I have to suggest, use your tuya plug for something else and find some
tasmota flashable device. Using tuya api is too complicated for something
that should be simple as a single web request.

Carlsans, 2025-01-22:
You can get tasmota plug on ali for about 15 Usd. Twice the price but if
you value your time and privacy, they are dirty cheap.....

JoAl115, 2025-01-22:
I am sorry I deleted my comment, I figured it out just after I post my comment (of course). Your plugin works, don't worry, issue was that when I first tried all options, plug version 3.4 didn't do anything, and when I tried again as last resort I gone trough all plug versions it finally started working under v3.4.

Thank you 👍

Carlsans, 2025-01-22:
Yes, that was supposed to be done automatically...